bangcorrupt/hacktribe #1: Repeatedly promt to reboot or start update

BKLronin, 2021-11-16:
Hi!

I changed the 4 postiins in hex like the script does and I can get to a successful boot with electribe sampler written on startup. From there it repeatedly asks to reboot or press enter. Grey synth es on 2.02

Does the patch solve this you provide solve this and is it confirmed working?

Thank you very much.

bangcorrupt, 2021-11-17:
Hi

Flash back to the factory synth firmware 2.02, then update with this patched firmware.

Hacktribe will run on either hardware, I've experienced no issues.

bangcorrupt, 2021-11-17:
See [Boots to prompt](../../wiki/debrick#boots-to-prompt) and  [Prepare SD Card](../../wiki/debrick#prepare-sd-card) in the wiki.

BKLronin, 2021-11-17:
Those are great news. 

I successfully came back to the standard firmware yesterday. Will flash the patched version today. Can´t wait and thank you for your effort.

BKLronin, 2021-11-17:
Totally worked. I´m impressed. Theres even sum very usable new synthesis options and all the stuff that is mentioned. This is EPIC! Noodled on it since the first one released and already had the blue one before so I was really tired of the sounds :)

Sorry for spamming the issues section, didn´t found no ther active place to write.

I skipped the checksum part cause I didn´t get it to work on manjaro or power shell.

bangcorrupt, 2021-11-18:
No problem.  

> I skipped the checksum part

I'll add a python script to handle checksum verification and patching automatically.

closed 2021-11-18 by bangcorrupt