bangcorrupt/hacktribe #2: Loading error when saving to factory banks

BKLronin, 2021-11-19:
Using the electribe sample editor I get reading error, when booting the electribe, as soon as I use one of the factory banks. The same sample works on the user bank.

Not a huge thing but maybe there is a trick to it.

bangcorrupt, 2021-11-19:
Can you save to the factory banks using the factory firmware?  

I didn't think this had ever worked, I can't test right now.

When using [Oe2sSLE](https://github.com/JonathanTaquet/Oe2sSLE) I always start at slot 501, you can set this in the "Import Options" dialog.

BKLronin, 2021-11-19:
I think not, I don´t have the sampler stock firmware.

501 to 999 works fine. Thats plenty of space.

I made a short video about the new features. Hope thats ok. https://youtu.be/n0wXUqgfa9Q

bangcorrupt, 2021-11-19:
I think this is the expected behaviour.  At some point in the future I'll change it.  

Video is totally cool, thank you.

bangcorrupt, 2021-11-19:
I linked to your video in the Readme, let me know if want crediting differently.

closed 2021-11-19 by bangcorrupt