bangcorrupt/hacktribe #3: Performing the Update with Windows and Sample Loading/Editing

johnforrest2021, 2021-11-21:
This is amazing and great work! Loaded this up on my e2 with no major issues. This makes the e2 operate exactly as I wanted it to. I have  a couple comments/questions on the procedure and a couple possible bugs I have found:
Note: I have a Windows 10 machine and am a novice to both GitHub and the e2

Procedure Notes:
1. When using Windows, I used CertUtil to perform the sha256 checksum. 
2. As a windows user i had to download bspatch from romhacking.net (https://www.romhacking.net/utilities/929/)
 and extract the entire contents to the "hacktribe" folder to be able to use it to apply the patch to the SYSTEM.VSB file.
3. The checksum was different for the SYSTEM and hacked-SYSTEM files, does this matter?
[image]

New Firmware bugs(?)/comments:
1. When loading/recording a sample, the e2 will speed up the sample. To get it back to normal, I either need to lower the speed/pitch by 20% or change the tone by -5 in the sample edit menu. (Not sure if this is a bug or if I need to read the user manual). This occurred both when loading a sample from a card and when I resampled one of my loaded samples. I have not tried recording a sample from audio in at this time.
2. When editing samples, the Time Slice function has an issue. Once the sample is sliced, you cannot preview the slices of the sample.
3. Is there a way to get the original e2 synth samples loaded as the default?
4. What happens if you perform a factory reset when the hacktribe firmware is loaded?
5. Will this firmware behave like the e2s and automatically try to load the all sample file from the SD card upon startup?

This is great work! Thanks for your effort.

bangcorrupt, 2021-11-21:
I don't know a lot about Windows.

> i had to download bspatch from romhacking.net

I found [bsdiff for Windows](https://www.pokorra.de/coding/bsdiff.html) which appears to provide [bsdiff and bspatch executables](http://download.pokorra.de/coding/bsdiff_win_exe.zip).  I haven't tried these.

>The checksum was different for the SYSTEM and hacked-SYSTEM

The checksums for each file should be different.  Each checksum for the flies you have should match the checksums saved  in the `hacktribe/hash` directory.  ( I just noticed this has been deleted at some point, will fix shortly).

>the e2 will speed up the sample

If you used the init pattern I provided it's probably set to the key of F.  Check the `Key` setting in the main pattern menu.

> Once the sample is sliced, you cannot preview the slices

I think this is how the factory firmware works.  I haven't intentionally changed anything with sample slicing.

>Is there a way to get the original e2 synth samples loaded as the default?

Not yet.

>What happens if you perform a factory reset when the hacktribe firmware is loaded?

Factory reset overwrites only the patterns and global settings.  If you had a factory synth you would get the factory patterns and global settings from the synth.  Firmware would remain the same.

>Will this firmware behave like the e2s and automatically try to load the all sample file from the SD card upon startup?

Yes.  This is a modified version of e2s firmware.  It has all features from e2s and most features from e2.

Thank you for testing.

bangcorrupt, 2021-11-21:
Hash files restored 2425dbf.

johnforrest2021, 2021-11-21:
Thanks for the quick response! 

Thanks for updating the hash files. 

For the sampling speed issue, what you said is correct. After I record the sample, it will play at fast speed when I am reviewing it. When I import it into a pattern and set the pattern key to C from F, it will play at normal speed. 

I have been able to change over my e2 to e2s using the 1.06 sampler firmware in the past. On that firmware, the time slice function allows the user to split up a sample into automatically cut parts. When you are in the Threshold menu you can set where the sample gets split, preview and select which one you want to use. The new firmware appears to not be able to do this.

Last question for now: if I put my new SYSTEM.VSB on an SD card in the proper directory, I can just load it on to another e2 without going through the rest of the procedure, correct?

bangcorrupt, 2021-11-21:
Ok, I'm not sure what's going on with slice previewing.  I'll look into it when I next work on this.  

>I can just load it on to another e2 without going through the rest of the procedure

Correct.

johnforrest2021, 2021-11-21:
This is great work, thanks!

closed 2021-11-21 by johnforrest2021