bangcorrupt/hacktribe #9: Exportig all samples causes error on start.

BKLronin, 2021-11-21:
When I use Export all samples after I recorded something and want to save it to the SD, it saves a bunch of drum samples to the factory section causing the known "an error occured" text on startup. When I remove them with the sample manager its booting up nomral again. 

[image]

bangcorrupt, 2021-11-21:
Did you import `samples/hacktribe-blank-e2sSample.all` after the first boot into hacktribe?

Starting with this file instead of the factory Sample.all file is supposed to fix that issue.

BKLronin, 2021-11-21:
I started clean and did an initial "Export All Sample" to generate a blank Sample.all at first.

bangcorrupt, 2021-11-21:
I think if you import `samples/hacktribe-blank-e2sSample.all` first, and then export all samples it should work.

BKLronin, 2021-11-21:
Copy, will try that.

bangcorrupt, 2021-11-21:
I may have been wrong about that, exporting all samples should have worked fine the way you did it.  

I can't reproduce the issue.  Can you give step by step instructions to reproduce the issue starting with a fresh firmware update?

BKLronin, 2021-11-21:
I will try, I can say yet that these are samples that I have in the slots 501-514. They get copied to 346 factory slots. Even if I export a track as multiple wave file to sd. 

Doesnt happen when I only export a single sample.

bangcorrupt, 2021-11-21:
Have you ever edited the Sample.all file with software?

Can you share a Sample.all file that demonstrates the issue?

BKLronin, 2021-11-21:
I used the Electribe Sample Manager to import the Drum samples after I made that clean Sample.all file.

I´m looking into smaller causes right now. I got tons of Drumshot samples on the SD mayber that caused something.

bangcorrupt, 2021-11-21:
[This Electribe Sample Manager](http://www.appification.de/ElectribeSampleManager/)?  I haven't used it, only  [Oe2sSLE](https://github.com/JonathanTaquet/Oe2sSLE).  Can you share a Sample.all file edited using Electribe Sample Manager?

BKLronin, 2021-11-21:
Yes that one.
[e2sSample.zip](https://github.com/bangcorrupt/hacktribe/files/7576831/e2sSample.zip)

Will try to repeat the error with this fresh file now.

bangcorrupt, 2021-11-21:
I booted with this Sample.all, recorded 2 new samples, exported all samples and rebooted without issue.

BKLronin, 2021-11-21:
Worked 1 time the second time it created sample 346 in factory sample area. Not related to the Sample manager it seems. HAppening after the export.

[e2sSample.zip](https://github.com/bangcorrupt/hacktribe/files/7576857/e2sSample.zip)

BKLronin, 2021-11-21:
Loaded your blank all file and recorded 3 samples than exported and all good.

Seems to be something that is introduced by the sample manager :)

bangcorrupt, 2021-11-21:
Excellent.  I've always used Oe2sSLE, it has a few bugs but as long as I start at slot 501 the Sample.all files work fine.

Thank you for testing.

BKLronin, 2021-11-21:
I always have problems with Tkinter under windows. It should be integrated in latest python but never works. Better on Manjaro.

I can confirm this for now. Edited the samples agfain after the 3 successful records and it introduces this "memory leak" right after recording they show up in factory.

Might make a single file runtime Version of the Oe2sSLE for ease of use.

bangcorrupt, 2021-11-21:
I noticed in the first file you shared, the first sample is in slot 502.  Does it make a difference if you start one slot before?

BKLronin, 2021-11-21:
Cant tell for sure. It worked when i didnt record audio. Now as soon as I do something with the sample manger it gets leaky. -Error unknown error. -

Cant reproduce exactly. Will try the othe manger now, if that works its safe. Everything else is more time consuming :)

BKLronin, 2021-11-21:
Strange, same problem when using the other sample manager. It happens on import already (switching on).

BKLronin, 2021-11-21:
So I tried a few things and when I leave 501 empty it imports correctly even from the electribe sample manager. When I know record a sample to 501, because I cant change the slot, its still fine but as soon as I record the next one it overflows to 346. New recordings get safed correctly but always create another entrie after 346 too. (Thats causing the error on reboot then)

I flashed the firmware again and it went well, I did factory reset once, accidentially. Maybe that caused some harm.

bangcorrupt, 2021-11-21:
Is this still happening if you start with `hacktribe-blank-e2sSample.all`?

I just thought, did you do the old firmware swap on this machine?

When I did this in the past it left some junk data on the flash after having the sample saving error.  I never found out if it was the cause of a bug I had , but it's worth checking.  I'll add a script to read/write flash via SysEx so we can get a better idea of what's happening.

BKLronin, 2021-11-21:
> Is this still happening if you start with hacktribe-blank-e2sSample.all?
Yes tried different methods, more than 5 samples -> ploblem

I did not tried the old swap but tried to modify the header myself before. Just the 4 symbols plus something in the korg at the very bottom. Restored succesfully to stock.

Sounds good. Couldn´t find a workaround. If I add samples and dont do anything on the machine it seemed to work thats why I found no issues at the beginning but sampling or adding samples in machine causes this.

bangcorrupt, 2021-11-21:
> more than 5 samples
Ok, I can recreate this, but only on one machine.  

On the machine I loaded your e2sSample.all files to I have the same issue.  

I tested on another 2 (one synth, one sampler) and neither have this issue. 

Can you share the original sample files you imported with the editor?

BKLronin, 2021-11-21:
Interesting. Mine is serial nr 005513. Maybe somthing with the early tribes?

I used different samples all the time. Doesnt seem to matter. Internal recordings cause the same. [kicks.zip](https://github.com/bangcorrupt/hacktribe/files/7577233/kicks.zip)

bangcorrupt, 2021-11-21:
My 2 synths are 20xx, one has the issue, other does not.  I think it was caused by something in one of the e2sSample.all files.  

My working theory is that the Electribe Sample Manager handled some wav metadata poorly, which led to some flash corruption, which is causing a residual issue.  Seems like a longshot.  

Tomorrow I'll add scripts to dump some parts of flash and RAM to remove the guesswork.

BKLronin, 2021-11-21:
Sounds good :)

johnforrest2021, 2021-11-22:
FYI - I am also having the exact same issue as BKLronin, but I am not using sample manager. I recorded and edited some samples on the device from 501-506. Sample 346 gets written with whatever was saved to 506. This issue occurred after I started it up with the SD card in it and it loaded my saved e2sSample.all file (with error). I did not run the clear sample file.   SN 001030.

bangcorrupt, 2021-11-22:
Ok, looks very likely this is due to not using the `hacktribe-blank-e2sSample.all` to clear all samples after first boot.  I think this is the same error as there was with the old firmware swap.  `hacktribe-blank-e2sSample.all` is the fix for that, but if it's not used first, a load of junk data gets written to the flash and causes an ongoing issue.  

This should be fixable once I implement #10, for now I'll update the Readme to make it more clear.

bangcorrupt, 2021-11-22:
@johnforrest2021 
Is your hardware a factory synth or sampler?

johnforrest2021, 2021-11-22:
I have the e2 synth. Will be able to reflash and run the blank sample file later today, will let you know how it turns out.

bangcorrupt, 2021-11-22:
This issue appears to be resolved by overwriting the PCM section of the synth flash with the PCM data from the sampler flash.  This data needs to be extracted from factory sampler hardware.  I'll add scripts at some point in the future.

>reflash and run the blank sample file

`hacktribe-blank-e2sSample.all` turns out to be irrelevant here.

BKLronin, 2021-11-22:
Ok cool, I checked the old thread and 523 should be the first safe slot. 

Is it possible to block the other slots or hide them somehow? That would make the sample function a little easier.

bangcorrupt, 2021-11-22:
>Is it possible to block the other slots or hide them somehow? 

Sounds like a good idea, probably the simplest fix.  I'll look into it next time I work on this.

bangcorrupt, 2021-11-23:
This is caused by a mismatch between the number of factory samples in the synth and sampler.

BKLronin, 2021-11-23:
Intersting, so it´s something like a duplicate written in the wrong register?

Why can´t they just open up all firmwares :D

bangcorrupt, 2021-11-23:
Fixed in 3d10cf9.  Doesn't appear to create any new issues.  It just sets the first available user sample slot to 523.  Any existing e2sSample.all files that use slots below 523 will need editing to start at 523 before importing.

Because samples below 523 were seen as factory samples, they were being saved to the flash and sending a bunch of stuff screwy (I didn't reverse the whole issue).

:

>open up all firmwares

The same sound engine is used in a few Korg products, KingKorg and MiniKP2s at least.  I was hoping they would release an SDK like for Prologue.

BKLronin, 2021-11-23:
Excellent, didn´t expected it to be solved so quick :) 

People always want more but I think those are still way more than enough sample slots. WiIl ltry this out asap. Reverteed to the opriginal synth firmware in the meantime with no issues. Even all the drum samples came back. Didn´t expect that. SO one can reverse 100% to synth it seems.

Yeah I think no matter how good a synth etc is, its just great when you can add a bit of fresh air from time to time. I´ve seen that on my minilogue xd as well. I almost liked the Digital slot more than the analog oscis :D Still it´s rarely seen. Microfreak has at least opened wavetable even if there is no sdk.

bangcorrupt, 2021-11-23:
>all the drum samples came back

The factory samples are all still in the flash, the firmware update doesn't overwrite them.  I don't _think_ this issue has caused any damage in that region, it only wrote metadata to an unused area of the flash (probably).

If hacktribe is booted without an e2sSample.all file, it will load the PCM data from flash.  They aren't referenced in the firmware, so they won't be available, but they will take up sampling time in the DSP RAM.  

This is what importing `hacktribe-blank-e2sSample.all` actually fixes.   You get the same result by booting without SD card, then immediately exporting all samples and rebooting.

BKLronin, 2021-11-23:
Thanks, I just wondered if they are included in the firmware file. I also understand now the importance of the clear bank. I´ll better upload the file before so its safer. 

So the other sample editor should be fine I guess?

bangcorrupt, 2021-11-23:
>sample editor should be fine

Yes.  I was way off the mark at the start of this thread.

BKLronin, 2021-11-23:
Finding bugs is tricky. Are the electribes where this doesnt happen then (The 506-523 hole)?

BKLronin, 2021-11-23:
Can confirm it working here. We still need to manually exclude samples from that area when importing. Got it. Sampling succesful takes 523 as first slot-and 524 on the next.

Good

johnforrest2021, 2021-11-23:
Was able to go back to the synth, FW 2.02, got all synth function/samples back. Then I performed a downgrade and transfer to sampler and back to synth with firmwares: 

E2_v2.02->E2_v1.03->E2S_v1.06->E2_v2.02. 

For the E2 Sampler v1.06 firmware:
1. When I used the cleared e2sSample.all file, it will clear all samples from 19+
2. Samples 501-522 will overflow into 346... just like with the hacktribe firmware.
3. When sampling using the electribe, it does not speed up the sample like I get with the hacktribe firmware.

After flashing back to E2v2.02, got everything back. Performed the new update from this morning. Now I am on hacktribe firmware. Ran the cleared e2sample.all file included in the repository. This clears only samples 510+.

When recording new samples, it will start at 523, but I still have the ability to load samples from the SD card in 501-522 with the overflow issue. I also have the sped up sample issue. @BKLronin do you get sped up samples when recording a new sample?

Thanks again for the effort, any updates you make I'm willing to test.

bangcorrupt, 2021-11-24:
>This clears only samples 501+

Are you seeing factory samples after the new oscillators?  Or does it jump from 274 to 501?

>sped up sample issue

I can confirm this.  See #12.

>still have the ability to load samples from the SD card in 501-522 with the overflow issue

I'll look into this

BKLronin, 2021-11-24:
@johnforrest2021 
When in the C Dorian scale its not pitched up for me.I checked with a guitar tuner.

johnforrest2021, 2021-11-24:
> Are you seeing factory samples after the new oscillators? Or does it jump from 274 to 501?

@bangcorrupt 
It jumps from 274-501. The factory synth samples are no longer present with the hacktribe firmware.

> When in the C Dorian scale its not pitched up for me.I checked with a guitar tuner.

@BKLronin 
When recording the sample and playing it back before saving it will be sped up. Once selected as an OSC in the pattern(C, Dorian scale), it will go back to the correct speed.

bangcorrupt, 2021-11-24:
> The factory synth samples are no longer present 

Ok, this is the expected behaviour.

See #12 and #15 for other issues discussed in this thread.

closed 2021-11-24 by bangcorrupt