bangcorrupt/hacktribe #16: Quick Transpose (Feature Request)

Shoontz, 2021-11-25:
On other Electribes you can quickly transpose a sequence in keyboard mode by holding shift and pressing a new key.
something like this on the Electribe 2 would open up ### real-time performance tremendously.

bangcorrupt, 2021-11-25:
This is one of the first things I wanted to add.  I haven't hacked any of the menus or controls yet, so no idea how difficult it will be to implement.  There is a lot of interaction between different processes when a control is changed, it's difficult to hijack at a single point.  I'm going to make a lot of things configurable via SysEx before working out the UI.  

For now, you may be able to do read-modify-write on the pattern file via SysEx without interrupting the audio.  Would definitely need to use USB MIDI, TRS is too slow.

Shoontz, 2021-11-25:
bangcorrupt, I could kiss you!  Thank you for all your hard work!

ecco00, 2022-01-03:
man, just use the KEY menu item to transpose the one you're in... i mean, it's even available as a shortcut via shift+numpad :D

Shoontz, 2022-01-04:
Hey ecco00, it's just not the same.  Having transpose work as it does on the older Electribes or the Beatstep would be such a killer real-time performance feature.  I think you'd probably have to use them to appreciate what a step backwards transpose is on the E2.

ecco00, 2022-01-04:
true dat, any example i should check out to better understand what you mean?

Shoontz, 2022-01-04:
You can see in this video how transpose used to work on the Electribes at about the 23-minute mark in this video.  :-)   https://youtu.be/O_3pnbTq26o?t=1436

ecco00, 2022-01-05:
the true difference i see is the menu-dive needed