bangcorrupt/hacktribe #17: Late IFX Value "quantized" after pattern reload

BKLronin, 2021-11-25:
When using the late effect on  asnare for groove purposes, it seems the it gets quantized later as the set value. The value itself is correctly safed and after setting to orig value the timing is correct again.

bangcorrupt, 2021-11-25:
Can confirm, will investigate.

bangcorrupt, 2021-11-26:
This may have to wait until I get a debugger attached to the DSP.  It looks like it's returning to the default value in the effect unit definition.  Will require stepping through to see how the DSP code handles the data sent from the CPU.

BKLronin, 2021-11-26:
Ok cool, I´m just making some beats currently to test all functions.

BKLronin, 2021-11-26:
Is it just me or is the recording quantization much finer? I can remember quantizing almost all of my grooves out in the original fw. (not IFX related)

bangcorrupt, 2021-11-26:
Shouldn't be any difference.  If you're not using any groove template or swing, everything should be grid locked.

BKLronin, 2021-11-26:
Ok, just placebo or my timing got more conventional recently :D

The lag ifx is pretty nice for that. LIttle more natural to dial in other than the grooves.

bangcorrupt, 2021-11-27:
It looks like all IFX are being reset to  Edit value of 64.  Probably missed updating a pointer somewhere.

BKLronin, 2021-11-27:
At least for lag and overdrive the orig value when loading the pattern is correctly safed but I have to manually move it to this position where it shows "orig". Thats not the case for one of the filterdrivers they have the 64 orig value as you have said.

bangcorrupt, 2021-11-27:
a939c6a appears to recall IFX Edit values correctly.

bangcorrupt, 2022-01-15:
1. Set IFX type to 'LP Drive'
2. Set IFX Edit to minimum
3. Save pattern
4. Increment pattern
5. Decrement pattern
6. Increment IFX Type
7. Decrement IFX Type

8. IFX Edit has returned to default 64 <-- Error

9. Increment IFX Type 
10. Decrement IFX Type
11. IFX Edit value is recalled correctly

Find missed index and set to IFX max count.

bangcorrupt, 2025-05-17:
Fixed for real this time. [3a64257](https://github.com/bangcorrupt/hacktribe/commit/3a64257d88bb10a5aa65d344b9b01bf79ac7d0f8)

closed 2025-05-17 by bangcorrupt