bangcorrupt/hacktribe #18: hacktribe on mk2 units

gradientledge, 2021-11-25:
can anyone confirm this wont brick the blue zinc unit, ive only seen people doing this on the grey and black models but the people mentioning trying this on the blue units brick their devices, so is there any confirmation of this working on the blue unit?

bangcorrupt, 2021-11-26:
>  confirm this wont brick the blue zinc unit

I can't test, I don't have one.  They run the same factory firmware.  As far as I know, the old firmware swap worked for all devices.  

> people mentioning trying this on the blue units brick their devices

Where are you seeing this?

gradientledge, 2021-11-26:
just on comments on threads ive read, not just this but also in threads that were about the os swap as well. im sure it works fine.  they might of just didn't complete the patch instructions correctly. i just thought id bring it up if it happens to be a problem later on great work though man once all the bugs are ironed out this is going to be awesome.

gradientledge, 2021-11-26:
did it this morning confirmed works on blue units no problems.

bangcorrupt, 2021-11-26:
Awesome, thank you for testing.

closed 2021-11-26 by bangcorrupt