bangcorrupt/hacktribe #27: Modify e2sSample.all file format to include key/velocity mapping.

bangcorrupt, 2021-11-28:


gradientledge, 2021-11-29:
i have 2 programs to edit the .all files but are full of bugs, and every all file pops an error when i try to load it any suggestions? i pretty much just put the blank in there and edited it from the device.

bangcorrupt, 2021-11-29:
> any suggestions?

Answered in #29.

BKLronin, 2021-11-29:
I´ve read somewhere that the factory bank of the Sampler uses at least velocity for theire samples.

bangcorrupt, 2021-11-29:
Looking at the data in the flash, the sampler doesn't use key or velocity mapping for any samples.  The synth uses key mapping for the keys, bass guitar and I think horns.  When a note is triggered, the key and velocity information is read from RAM, but for most samples there is only a single key group and velocity layer.  The existing data format allows for 4 velocity layers and (maybe) 128 key groups, but there is not a section for this information in the e2sSample.all file format.

BKLronin, 2021-11-29:
OK, so there is no possibility to save the layers or zones? Maybe they ditched during development for some reason, like eating up voices as one of the layer is still playing when triggerin the other. That would cause like 4 voices per part when all velocity zones are used. 16 for a small drumset is a lot :D

BKLronin, 2021-11-29:
What about embedding multiple velocitys, from a bass guitar for example, into a loop and map the individual slices to the velocitys? That would trigger just one voice too. Maybe use the "tune" variable to define the slice position for example. 

Might be really difficult and I´m not sure if its worth it.

BKLronin, 2021-11-29:
Ah ok, I thought its implemented but cannot be used as the "all" file cant save it in any way.

bangcorrupt, 2021-11-29:
I think I can add to the end of each sample block, or put a new section at the end of the file.  Will definitely be a long time.

BKLronin, 2021-11-29:
Hijacking the tune parameter might be easier because the editors can be used unchanged and maybe fewer internal problems?

Might be something for later anyway.