bangcorrupt/hacktribe #30: Import/Export Grooves/FX via SysEx.

bangcorrupt, 2021-11-29:
Add feature to import and export groove templates and FX presets via SysEx.

gradientledge, 2021-12-05:
so is there a way to transfer gate arp patterns as well its sucks they have no triplets or just useable patterns.

bangcorrupt, 2021-12-05:
I haven't looked for the arp patterns yet.  If I can find them, we can change them.

gradientledge, 2021-12-06:
so for the sysex are you just using midiox to communicate with it, or is there a better program as that ones ancient?

bangcorrupt, 2021-12-06:
> are you just using midiox to communicate with it

Answered in #37.

bangcorrupt, 2021-12-30:
Get/Set IFX added in b700ac9.

bangcorrupt, 2021-12-30:
Get/Set grooves added in 070b282.

DepreTux, 2021-12-31:
> Get/Set grooves added in [070b282](https://github.com/bangcorrupt/hacktribe/commit/070b282123f87812977e1ede9c8bda226a4fb07d).

How does the "add ifx" option differ from the "set ifx" one?
Does it mean we can add IFX to the list?

Thanks!

bangcorrupt, 2022-01-01:
Yes, `add_ifx` adds a new IFX preset to the top of the list, increasing the total count.

DepreTux, 2022-01-02:
> 

Amazing... I'll give it a go as soon as I can figure it out!
Thanks!