bangcorrupt/hacktribe #71: IFX edit Sequence stop note on problem E2

richardarnezjames, 2022-01-12:
Switching IFX type or editing parameters sometimes causes voices to stop sequencing but keeps note-on for all the voices that were unmuted. Switching sequences sends all controller reset and allows playback but the issue keeps happening.

bangcorrupt, 2022-01-13:
I can't reproduce this.  I tried playing busy patterns and switching the IFX type and edit parameter like crazy.

Can you share steps to consistently reproduce the issue?  If it happens with specific patterns or samples, please share the files.  If you can't do this, a video of the issue occurring is useful.  

Does this only occur with hacktribe, or is it also an issue with the factory firmware?

Thank you for supporting hacktribe! :smiley:

richardarnezjames, 2022-01-13:
I'll do that this weekend, first I need to revert my firmware back to the E2 Synth for a rehearsal tonight. I thought I understood how to do that but I don't. Haven't bricked it tho so gimme the deets!
 
And thank you for this sick hack. I enjoy it I just need to be able to revert back atm.

bangcorrupt, 2022-04-01:
I think I faced the same issue.  Does this pattern demonstrate what you were experiencing?

[issue_71.e2spat.zip](https://github.com/bangcorrupt/hacktribe/files/8401203/issue_71.e2spat.zip)

The pattern has one note per step for every part, oscillator 'SAW'.  Part 1 has oscillator 'DUAL-SAW' (added from the synth firmware).  

To reproduce the issue, play the pattern and scroll quickly through MFX types. If the sequence does not hang, set more parts to 'DUAL-SAW' oscillator and scroll MFX again.  I find it is more likely to hang at lower BPM, or with analog modeled filters active.  

At a guess based mostly on assumption, it may be something to do with voice management.  The new oscillators are in slots where samples should be, and may be reporting as requiring only one voice while actually using two.

bangcorrupt, 2022-04-01:
I think I faced the same issue.  Does this pattern demonstrate the issue you were experiencing?

[issue_71.e2spat.zip](https://github.com/bangcorrupt/hacktribe/files/8401276/issue_71.e2spat.zip)

The pattern has one note per step for every part.  Parts 1-8 have oscillator 'DUAL-SAW' from the synth.  Parts 9-16 have oscillator 'SAW'.

To reproduce the issue, play the pattern and scroll quickly through MFX types.  If the sequence does not hang, set more parts to 'DUAL-SAW', reduce the BPM, or enable analog modeled filter types.  

At a guess based mostly on assumption, this could have something to do with voice management.  The new oscillators are in slots where samples should be and may be reporting as only requiring one voice, while actually using two.

bangcorrupt, 2023-02-28:
Fixed in [6f48288](https://github.com/bangcorrupt/hacktribe/commit/6f4828832eb8b1cc04f93965c143fc4c45cc2ab5).  There are still some timing issues when scrolling through oscillators and fX, but these are also present in the factory firmware.

closed 2023-02-28 by bangcorrupt