bangcorrupt/hacktribe #103: Building bsdiff 4 fails on Windows

bangcorrupt, 2022-01-16:
#91 

[GUI Testing](https://github.com/bangcorrupt/hacktribe/discussions/97#discussioncomment-1979708)

[bsdiff4/issues/20](https://github.com/ilanschnell/bsdiff4/issues/20)

BKLronin, 2022-01-16:
I can test this tomorrow if it's just an admin issue. I got python 3.10 already installed. I could build the pyinstaller version as well if desired.

bangcorrupt, 2022-01-17:
#91
@johnforrest2021
> I was able to use pip installer to get argparse bsdiff4 on my Windows 10 machine. After installing visual studio 2022 that comes with C++ 17.0, I ran the pip install command shown in the tutorial and it works.

johnforrest2021, 2022-01-17:
> 

Just a clarification I am using anaconda prompt for this. 
[image]

bangcorrupt, 2022-01-17:
Does it fail if you use use standard command prompt?  

I don't know much about Windows, my understanding is that anaconda prompt adds some directories to the PATH variable.

johnforrest2021, 2022-01-17:
> Does it fail if you use use standard command prompt?
> 
> I don't know much about Windows, my understanding is that anaconda prompt adds some directories to the PATH variable.

For all the python parts of this hack I use anaconda prompt, I don't have pip installed on my machine in such a way to be able to use it with normal command prompt. Maybe @BKLronin might have more insight?

bangcorrupt, 2022-01-17:
Ok, no problem.  I'm setting up a Windows 10 VM, hopefully I can compile an executable with no dependencies.

BKLronin, 2022-01-17:
I don´t,  usally I do all my python stuff in manjaro. I do have pip installed and it worked but I didn´t want to install this huge VSStudio package as I have VScode already installed. VC redist 14.30 didn´t worked. All with powershell.

bangcorrupt, 2022-01-17:
> VC redist 14.30 didn´t work

It didn't work with VC Redist 17 for me either.  

It did work after installing Visual Studio 2022 with Desktop Workload.  I'm not sure which individual components it required from the >8GB installation.

closed 2022-01-23 by bangcorrupt