bangcorrupt/hacktribe #170: SD Card Capacity Error:

redwallx140, 2022-06-10:
After loading Hacktribe, unit is much more picky about card size. It was previously able to use cards up to 64GB (ptentially out of spec but working). I had to drop to a 4GB card for immediate music work. Will test others. Can this be resolved?

bangcorrupt, 2022-06-10:
Can you share more information about the error?

What happens when it doesn't work?

Does the same card work with factory sampler firmware 2.02?

How much data is on the card and what is the layout?

Have you formatted the card using hacktribe?

redwallx140, 2022-06-11:
I tried a few cards. All worked up till 64gb sdxc an up. I formatted them fat32 and the device wouldn’t use the card. Tried blank card. Will try making the KORG folder myself and see if it reads. Otherwise, I can stick to smaller cards. No big deal and can work on it over time.

redwallx140, 2022-06-11:
Also am testing with other E2

bangcorrupt, 2022-06-11:
It should only support SDHC up to 32GB.

What happens when it doesn't work? Is there an error message? How does it fail?

redwallx140, 2022-06-11:
The error is “card error”. It was a 64gb card. This must be the problem. Will use 32 and under. I wasn’t aware of this limit

closed 2022-06-17 by bangcorrupt