bangcorrupt/hacktribe #283: Motion recording not working

tripsylon, 2023-09-20:
Hello, after the latest update the motion recording isn’t working on numerous projects, even after saving the project on another pattern number and after rebooting the machine… I’m using the red ES2.

bangcorrupt, 2023-09-20:
#284

Steps to reproduce:

1. Initialise pattern
2. Select 'Tap Delay Sync' MFX
3. Attempt to record motion sequence

Result:
Record light will not blink and motion sequence will not be recorded.

This looks like it is due to an audio effect being in a slot where a sequencing effect is expected ('Tap Delay Sync' replaced 'Seq Double').  Moving 'Tap Delay Sync' to a different slot may solve the issue.  I will test this later.

bangcorrupt, 2023-09-21:
Fixed.  'Tap Delay Sync' now replaces 'Touch Wah' in slot 21.

closed 2023-09-21 by bangcorrupt

tripsylon, 2023-09-25:
Working fine now 👌. Thank You 🙏🙂