bangcorrupt/hacktribe #298: Card format error and cant import samples

Gurthang19, 2023-11-21:
Recently i got an Electribe synth with the hacktribe already installed on it. There was no memory card inside. I tried to format several memory cards with the electribe and keep getting "card format error" message, but in the memory card the folder "KORG" is created. Is this a problem or i am missing something ? Tried importing samples but with no success.

bangcorrupt, 2023-11-21:
It sounds like you have an old version of hacktribe installed.  There were some bugs with card formatting when I changed the path from `KORG/electribe sampler` to `KORG/hacktribe`. (#147, #149)

The easiest way to fix it will be to update the firmware, using both SD card paths to make sure we get the right one.

1. Use the [Hacktribe Tutorial](https://github.com/bangcorrupt/hacktribe-tutorial)  to get the firmware update files.
2. Copy `sampler-hacked-SYSTEM.VSB` to `KORG/electribe sampler/System/SYSTEM.VSB`.
3. Also copy `sampler-hacked-SYSTEM.VSB` to `KORG/hacktribe/System/SYSTEM.VSB`.
4. Run the firmware update on the device and reboot.
5. Backup your SD card and format it on hacktribe to create the correct directory structure.
6. You can now delete the `KORG/electribe sampler` directory, only `KORG/hacktribe` will be used.

> Tried importing samples but with no success.

Try again after updating the firmware, also see [FAQ: I can't import my samples](https://github.com/bangcorrupt/hacktribe/wiki/FAQ#i-cant-import-my-samples).

bangcorrupt, 2023-11-30:
Assuming fixed as no  further comment.  Feel free to add more if you are still having issues.

closed 2023-11-30 by bangcorrupt