bangcorrupt/hacktribe #335: Shift+scroll sample doesn't jump over categories for Sample, Loop, User

vananasun, 2024-09-09:
Not sure if this is intentional. I have been renaming the category names for my sample pack and encountered this issue.
 Only happens on these categories and it is annoying!
Sample
Loop
User

bangcorrupt, 2024-09-10:
I think this is the same issue as #13.  I haven't looked into it yet, but it's probably because I changed the names of some categories.

bangcorrupt, 2024-09-15:
If you edit the category names in Hacktribe to match the factory sampler firmware, does it skip over them correctly?

It could be related to samples being in slots where they're not expected.

vananasun, 2024-12-29:
I tested it and figured out that it really is Sample,Loop,User categories with this problem. I reckon this was done on purpose to slowly scroll past break subdivisions and the like, but it is kind of annoying since you cannot rely on shift+scroll to select an entire different category anymore.

bangcorrupt, 2024-12-30:
I think you're right that it's a feature and I agree that it's annoying.  I noticed that some categories are skipped fully when scrolling backwards, but not when scrolling forwards.  That's definitely a bug. I haven't checked if this happens with the factory firmware or if I introduced it in Hacktribe. #13 is a separate issue.  

Fixed in https://github.com/bangcorrupt/hacktribe/commit/704dd7f2e6e7f3204f97479a9db3438bddf1ac4d.

I changed a conditional branch to always branch, skipping the code that treats those categories differently. I only tested briefly, but skipping those categories appears to work and nothing appears to be broken.

Thank you for sponsoring Hacktribe! :smiley:

commit 704dd7f2 (2024-12-30): fix(patch): Always skip full oscillator category, fixes #335.
closed 2025-01-01 by bangcorrupt