bangcorrupt/hacktribe #341: Sample Export Fail: Repeatable

OSCms20, 2024-11-04:
Hello, I have lost quite a few sessions worth of work, and have narrowed down the issue to this:

when pattern is playing, sample memory is in active use. In some cases, when play back stops, DSP continues processing audio in a way that impedes sample saving and causes error. A fix might be to issue an all-note-off before saving - thanks for listening.

bangcorrupt, 2024-11-04:
Hi, can you share steps to reproduce?

OSCms20, 2024-11-06:
I will verify version, but this type of crash has only happened in hacktribe. However I push it pretty hard :)  Basic steps are: record samples, load into parts, modify OSC start a bit, add TRiOSC mod, do some more that with samples IFX that have trails. Then, hit stop. as samples ring out, go to menu and data utiil export all.   

One thing to note, I navigate the menu really fast: Using SHIFT-PAD 16 ESC, I quickly get to the data utlity and perform function. I learned this to be the fastest way to get into the core utlities, and recognize that this might be too fast. 

As an aside, I also found out that if a light expander is applied to samples for import, all the OSC, IF and mods have consistent ability of push into self oscillation. This can be good or bad, but given the amount of modulation poasible I have printed some wild resamples and PTN exports doing this. Needless to say I am a power user.

OSCms20, 2024-11-06:
My family now recongizes my deflated grown at 11:45 on Saturday night when a fruitful sample session disappears into KORGpse.

bangcorrupt, 2024-11-08:
I haven't been able to reproduce this.  My hunch is that you're using an older version of Hacktribe and this is another symptom of [this bug](https://github.com/bangcorrupt/hacktribe/issues/71#issuecomment-1086404148).  If you update to the latest commit of the main branch and still face this issue I will look into it further.

bangcorrupt, 2024-12-30:
Closing as no further comment.  Feel free to add more information if the issue persists.

closed 2024-12-30 by bangcorrupt