bangcorrupt/hacktribe #365: Samples mislabeled as FM and VPM

awchx, 2025-03-22:
After adding the modified e2sSample file to the SD card, with samples starting from the number 501, I've noticed that some of them are mislabeled. Namely:

747-753 labeled as FM
873-903 labeled as VPM

(the numbers are base+482 to align with 501 as the starting point)

I've looked into the manual and correct labels seem to be:

265-271 (747-753 after mod) - labeled as SE
391-421 (873-903 after mod) - labeled as PCM

The Oe2sSLE editor marks them correctly too, in fact FM and VPM aren't even options to choose in there.

[image]

vananasun, 2025-03-30:
Yeah hacktribe has a few categories renamed:
```
(original)  (hacktribe)	
User	    Wave		
PCM	    VPM		
Loop	    Loop	
Phrase	    Sample		
Perc	    Perc.		
Tom	    Tom		
FX	    FX		
SE    	    FM		
Voice	    Voice		
Shots	    Instr		
Hits	    Hits		
Cymbal	    Cymbal		
HiHat	    HiHat		
Clap	    Clap	
Snare	    Snare		
Kick        Kick		
Analog	    Analog	
```

bangcorrupt, 2025-04-15:
When I changed the category names I was hex-editing the binary, so I replaced names of equal length.  I think I can add new categories at some point, maybe even make this configurable.

I'm going to call this a feature request and move it to Discussions.