bangcorrupt/hacktribe #369: SawUpB modulations dont work

VerKORGSt, 2025-04-15:
Since i installed hacktribe to my e2 sampler all of the SawUpB modulations dont work anymore

bangcorrupt, 2025-04-15:
I can't reproduce this, I tried:

- Patch latest hacktribe using Hacktribe Tutorial codespace.
- Install Hacktribe
- Test SawUpB routed to all possible destinations

Result:

SawUpB modulation works as expected.

Can you double check you are using the latest Hacktribe, patched using the Hacktribe Tutorial codespace?

If you still experience the issue, please provide steps to reproduce.

VerKORGSt, 2025-04-15:
Well i had the newest Version i Downloaded it now again and Updated it now it works

closed 2025-04-15 by VerKORGSt

bangcorrupt, 2025-04-15:
Can you provide a pattern file that demonstrates the issue?

bangcorrupt, 2025-04-15:
Turn up the Modulation Depth knob.  The default depth and speed is different for each modulation type.  For SawUpB the default depth is 0.

VerKORGSt, 2025-04-15:
Okay thank you Bro and i was wondering why it didnt work 😂