bangcorrupt/hacktribe #388: "Error no card" when booting with sd card installed

dev-mindz, 2025-06-29:
When i boot the machine with an sd card installed i see the hacktribe logo (no loading bar). this takes about 1 minutes 30 seconds to boot.

When going to data util to import sample or format or do anything card related, when opening for example import sample it says "Error no card". But it shows the sd card symbol at the top.

When i boot the machine without an sd card installed it boots in seconds. Then pressing the sd card in when booted i can go to import sample and it works fine.

Whats happening here? Like its working fine from that point till i turn it off again. then when booting i have to take the sd card out again, wait for it to boot. Put it back in and reimport all samples.

I have the gray/blue synth machine. followed the instructions in the [hacktribe tutorial](https://github.com/bangcorrupt/hacktribe-tutorial)

hacktribe is working without problems besides that

vananasun, 2025-06-30:
Nothing comes to mind, which SD card brand, size etc are you using?

dev-mindz, 2025-06-30:
Damn it might actually be the SD card. I found a micro sd adapter with a 2gb card in it. Copied the KORG folder over and now it works.

closed 2025-06-30 by dev-mindz